kp-bit/xblinds #4: [bug?] power consuming/stepper warm up over time

m00x-pixel, 2021-04-02:
Thank for for the current state include the switch.
I took some time to build first with breadboard, i want to check first before build all thing and i look to the power consuming.

After first start of the D1 Mini Clone will get 0.08A so around 0.4W  but after first turn it get up to 0.28A and 1.37W. 
From then it will stay at this point even it it is running or not, it seems the IN1 and IN2 from the ULN2003 is permanently on and get signal.
When i hit the reset button from ESP D1 will go back to 0.08A.

Is it possibel to set all pins the Pins to low after the work from the stepper is done, so it will not heat up also after?

kp-bit, 2021-04-03:
Good catch! I'll investigate...

kp-bit, 2021-04-03:
Easy fix, version 0.6 is now uploaded, thanks for reporting!

closed 2021-04-03 by kp-bit