kp-bit/xblinds #9: Cannot connect xblinds to home wifi.

maciak, 2021-04-14:
Hi,
I have flashed my D1 mini with v0.6 and it works fine with the setup I can connect to it's AP and open it's setup page but when I enter my wifi details it will not connect to it.
Terminal says only this:
```
[21:51:34]xBlinds
[21:51:34]dom
[21:51:34]my_wifi_password
[21:51:34]on
[21:51:34]
[21:51:34]1883
[21:51:34]
[21:51:34]
[21:51:34]xblinds/all
[21:51:34]xblinds-7c9fa2
[21:51:34]
[21:51:34]0
[21:51:34]0
[21:51:34]0
[21:51:34]0
[21:51:34]0
[21:51:34]
[21:51:34]Connecting to dom
```
Please let me know what can I do to properly connect this device to my home wifi.

kp-bit, 2021-04-15:
It looks like you either input the wrong password or your home wifi is not 2.4 GHz - either way it should fall back to broadcasting the AP (maybe after a reset) if it doesn't connect, so you can input wifi details again.
If it doesn't, you'd need to clear the flash memory of the D1, this guy has a solution for that: http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

coolerkerl, 2021-04-15:
Here the same! Can it be the password length? It always works in my guest WiFi. It's physically the same wifi

kp-bit, 2021-04-15:
Do you see the correct password in the terminal, even after a reset? The variable is "char password[32];" which I figured would cover most usage...

maciak, 2021-04-15:
Hi, Yes I can see the proper password. It has 11 characters so not that long. It looks like it hangs after showing message Connecting to "SSID". It does not show the AP anymore and behaves the same after every reboot. I've also tried different ESP2866 boards and on all it looks the same.

coolerkerl, 2021-04-18:
For me, the password is too long. I have more than 32 characters. 48 to be exact. I use a shorter one in my guest WiFi. But I won't be able to use the WiFi. Thanks for the nice project

closed 2021-04-29 by kp-bit