kp-bit/xblinds #13: Suggestion: use power supply from D1 Mini USB port?

MagaliDB, 2021-04-18:
Wouldn't it be easier to use the power supply from the D1 Mini USB port? I tested the assembled unit with the D1 Mini connected to my laptop with USB, and it seems to work just fine that way. I was able to control the unit.

arthurkok2, 2021-04-18:
This is what I have done as well. I actually modified the STL and added a hole on the side for the USB port. A laptop USB port might not deliver enough current, better to switch to a 2A switching USB power supply.

MagaliDB, 2021-04-18:
> This is what I have done as well. I actually modified the STL and added a hole on the side for the USB port. A laptop USB port might not deliver enough current, better to switch to a 2A switching USB power supply.

Could you upload the modified STL as a remix on thingiverse? Would love to use it, that way I don't have to use my soldering iron to burn a hole on the side haha

arthurkok2, 2021-04-19:
> Could you upload the modified STL as a remix on thingiverse? Would love to use it, that way I don't have to use my soldering iron to burn a hole on the side haha

here you go! https://www.thingiverse.com/thing:4832349

kp-bit, 2021-04-19:
You can't pull enough current from the 5V pin when powering the D1 from USB to provide any kind of torque on the stepper, and you can damage the D1 from trying. So you can power the D1 with USB but you would need to feed 5-12V directly to the ULN2003 (shared ground) for it to work - so it would be much easier to just power both with a 5V source, thus my design...

arthurkok2, 2021-04-19:
yea @kp-bit you might be right. I am reviewing the schematic - looks like there's a fuse and diode between the USB 5v and the 5V pin of the board.  So powering the motor through this method is definitely not idea as the fuse can blow.

[image]

kp-bit, 2021-04-19:
Not sure how accurate those data sheets are or if they fit all D1's, if there's a diode there, we shouldn't be able to power it via the 5V pin...

arthurkok2, 2021-04-19:
you'll still be able to power the board thru the 5V pin as the "+5V" rail is common. the diode is there to prevent the 5V supplying voltage back through the USB's VBUS

closed 2021-04-20 by kp-bit

MagaliDB, 2021-04-25:
@kp-bit you're right, I didn't read the documentation correctly. I thought you meant not to power it over USB from the pc, totally misunderstood that haha 😅 English is not my native language. 
It does indeed not pull properly when powered on USB only, so today I soldered a 5V 2.5A PSU (didn't have anything lower then 2.5A) and it works great!
Sorry for my misunderstanding and thanks again for this wonderful project.

kp-bit, 2021-04-25:
@MagaliDB, good to hear that it worked out for you - enjoy!