kp-bit/xblinds #14: More info on MQTT connection?

lexucd, 2021-04-18:
Hi,  I have made two of these xblinds for my living room and now i;m stuck on the MQTT part... don't seem to understand how it works exactly... i have made a https://node02.myqtthub.com/ account but i can only find Domain, clientID, add device...
no "MQTT group" so i don;t know where to subscribe...
please help!

ardenking, 2021-04-19:
Like most people I use a local Mqtt broker but it will work the same, you need to point the xblinds clients to the broker using address and credentials

kp-bit, 2021-04-19:
It should actually read topic rather than group, will correct that in next version... So just input server, port and credentials and whatever topic you'd like to use. Default is xblinds/all - as ardenking says, a local broker makes life a lot easier...

closed 2021-04-19 by kp-bit

lexucd, 2021-04-19:
I have tried to do that but whenever I enter those details, the d1 stops reacting and after reset it blinks once long and then it shuts down and can't be reached even from local ip...:(