kp-bit/xblinds #18: Settings surviving reboot

IndeterminateName, 2021-05-02:
Hi There,
Is there a way to have the D1 retain it's settings after reboot? I'm setting it up though the GUI and it is working perfectly but I'd like to be able to have it plugged in with the rest of my AV equipment which all turns off when not in use, I'd just have Home Assistant turn the plug on a while before it is scheduled to open or close and then turn off again when it is done. Is there a way to get the source code and bake the wifi, mqtt and calibration settings into the flashed file?
Cheers
Michael

kp-bit, 2021-05-02:
The D1 will retain its presets during a power down/reboot as they are written to flash memory when set. And if you make sure to use retain on your MQTT commands from HA, xBlinds will just pick up the latest command when powering on. Make sure you use v0.7 as it has a rewrite of the MQTT implementation. Source will be released eventually, when cleaned up and documented properly.

closed 2021-05-02 by kp-bit

IndeterminateName, 2021-05-02:
Hi,
In that case my issue is that settings do not survive a reboot when the intended functionality is that they should.
Cheers
Michael

kp-bit, 2021-05-02:
What version are you using? And what does the debug page say before and after a reboot?

IndeterminateName, 2021-05-04:
I was on 0.6 but I was also powering it via USB. I've now updated to 0.7 and powered it properly and it is now working.
Thank you

kp-bit, 2021-05-04:
Good to hear, enjoy!