kp-bit/xblinds #20: Hardware Hack - more powerful 28BYJ-48 - necessary code modification

spitzlbergerj, 2021-05-09:
Hi, I found a [hardware hack](http://www.jangeox.be/2013/10/change-unipolar-28byj-48-to-bipolar.html?m=1) to give more power to the small stepper motor. However, this must then be addressed via another controller. Instead of ULN2003 it is then a L293D, which is also available in small designs.

I would be happy to integrate this controller into your code if you make the source code for it available to me. If possible, I would make it so that the controller type can be selected via the configuration.

Thanks already, if you let me help.

kp-bit, 2021-05-09:
@spitzlbergerj, I actually don't think it would need code modification. The L293D is also controlled by 4 pins in same sequence as I'm using for the ULN2003, so it should just be a matter of wiring it up...

Defining code used in xBlinds for the AccelStepper library:

#define motorPin1  D1                       // IN1 on the ULN2003 driver
#define motorPin2  D2                       // IN2 on the ULN2003 driver
#define motorPin3  D3                       // IN3 on the ULN2003 driver
#define motorPin4  D4                       // IN4 on the ULN2003 driver
#define STEPS 2048
#define MotorInterfaceType 8

closed 2021-05-09 by kp-bit

spitzlbergerj, 2021-05-09:
Hi,
OK, thanks for pointing that out. I am waiting for the components and will try. After that I will describe my procedure and send you a pull request for a readme change if it works

praun, 2023-11-20:
> Hi, OK, thanks for pointing that out. I am waiting for the components and will try. After that I will describe my procedure and send you a pull request for a readme change if it works

Hi, did you ever manage to do this, if so I would like to know what it takes?