kp-bit/xblinds #21: More torque

lexucd, 2021-05-22:
Do you think that the L293D motor driver could be added instead of the 2003 for more torque? I saw something interesting here :https://lastminuteengineers.com/stepper-motor-l293d-arduino-tutorial/
Or even turning the 28BYJ-48 bipolar might do the trick for heavier rolls. 
But i guess the libraries would need to change...? (advanced noob in arduino and electronics)

P.s. Amazing work you are doing here!!!  Thank you a thousand times for myself and others that did not take the time but used your work.

kp-bit, 2021-05-22:
Hi, take a look at this issue: https://github.com/kp-bit/xblinds/issues/20 - you won't necessarily need other libaries if the other driver is connected to 4 pins...  Thanks for your input, enjoy!

closed 2021-05-22 by kp-bit

lexucd, 2021-05-22:
Tried it but it only turns clockwise when I try to set it up with + and -.
And also, after I enter the MQTT details, the device cannot be accessed from the local ip anymore, i have to re-flash it to enter the local ip again. (not that i've mannaged to make it work with cloud MQTT anyway - node02.myqtthub.com )

kp-bit, 2021-05-23:
Are you using the ULN2003 or the L293D driver? If it's the ULN then you might have a defective driver/stepper or faulty wiring.
As for the MQTT server, you need to use an IP address, there's no DNS client in the code...

lexucd, 2021-05-23:
I am trying the L293D now. 
 They work great with the  ULN2003 but I would like a bit more torque as it skips some "beats" when opening

kp-bit, 2021-05-23:
You might be wiring it wrong... Try setting up a piece of code with the AccelStepper library with the correct pinout and see if it works as intended.