kp-bit/xblinds #22: ESP8266 lässt sich anpingen, Website nicht erreichbar.

guber-a, 2021-05-28:
Hello, first of all great praise, great product what you programmed.
I have the following phenomenon after I controlled the XBLIND for the first time via MQTT, the web interface is no longer accessible. I see the blind in the network and can ping them too. Only when I start the MQTT broker can I access the web interface. Would it be possible to program the free inputs as up, down and stop. If you have called up, the motor will run until stop is pressed and the same in the other direction. Then I can control the blind regardless of the network or MQTT.

kp-bit, 2021-05-31:
The web interface should be accessible at all times, even when the unit is connected to MQTT. I think your problem here is that your MQTT server isn't always on? I'll look into how xBlinds reacts to the MQTT server disappearing, I could suspect that it spends too much time looking for it and doesn't service the web interface, it's on my todo list now. A workaround for you could be to not configure the MQTT option in xBlinds.
As for the "run until stop is pressed", that's really not a direction I want to go in as it sounds like a recipe for disaster to either stepper or blinds gears...

closed 2021-05-31 by kp-bit

guber-a, 2021-06-01:
Thank you in advance for replying.
This is actually how you write. After setting up the MQTT server, I was only able to access the web interface when the server is connected. If the server is switched off, the button does not work either.