kp-bit/xblinds #23: Shematic and Powersuply connection for 3 units

mikeo999, 2021-06-06:
Love your project and will start with it soon.

I have three windows (80 cm x 175cm) and want to use your project for this.

Would it be possible to connect alle 3 units to the same powersupply so i don't need 3x wirring and seperate power supplys to all units but combine them so i can use 2 main power wires that i can connect to all 3 of them. 

Any recommendation for a powersupply the feed all 3.

kp-bit, 2021-06-06:
Shouldn't be a problem, I'm using this PSU for four windows: https://www.amazon.de/gp/product/B07DQKM9P7 - however, I do run separate wires to each window (in a canal) but with the right wire dimensions you should be able to combine them.

closed 2021-06-06 by kp-bit

mikeo999, 2021-06-07:
Thanks for that. One final question.   

Is it possible to join them all so when you whant to close them it is juist hitting one button and the all go down or up at the same time with your software or is that only possible with extended software like HA.

kp-bit, 2021-06-07:
Yes, you would need MQTT for that and subscribe them all to the same topic. That's how I run it...

mikeo999, 2021-06-07:
Thanks wil try to search for more info how to do that.