kp-bit/xblinds #26: MQTT status in detail

LeoVe, 2021-08-18:
Hi,

Thanks for the great work!

I have v0.6 now fully integrated in my OpenHab2 installations. OpenHab2 uses the open/closed/half/full mqtt status info to display the status in de UI.

I noticed this was changed in v0.7/0.8.

Would it be possible to make the behaviour like open/closed of full detailed status info selectable in setup? So it would be able the upgrade to the next and possibly future versions for also the OpenHab users?

thanks!

kp-bit, 2021-08-18:
Hi @LeoVe ,

As I'm only running Home Assistant myself I wouldn't have the ability to test OpenHab2 functionality, so I don't really see a way to implement it in my code at this time.

However, I plan on releasing the source code along with v1.0 of the binary - probably some time during Fall 2021, with that you can make any changes that would make it work in OpenHab2.

Hope that helps...

closed 2021-08-18 by kp-bit

LeoVe, 2021-08-18:
Hi kp-bit,

The testing needed here is if mqtt status is back to the all modes open - close - half - full instead of open - close.
I think you won't need OpenHab for that.

If you make it configurable with a switch in the settings panel, both worlds would be happy.