kp-bit/xblinds #27: Can't connect to wifi network

mitchcecg, 2021-09-02:
**_I have less than a 50/50 chance of one of these being able to connect to my wifi network after entering the ssid and password after flashing. It works on about 30% of the d1's I purchase the rest do not work. However there is a difference in the terminal screen info that might help you figure out the ussue (see the 3 terminal screens below and my comments between them.
The Terminal screen shows (cr-lf removed)_** 
-----------------
Reading stored values from EEPROM...
xBlinds
on
1883
xblinds/all
xblinds
0
0
0
0
0
0
512
Turning on AP
WIFI_AP
Setting soft-AP ... Soft AP ready
Soft AP configured
Soft AP IP address: 4.3.2.1
HTTP server started
Setup ended!
---------------------------------------
**_It then shows this after entering the wifi info_**
_________________________________________
Setup ended!
Default
Default
Updating EEPROM...
Confirming EEPROM...
xBlinds
cecg_private
******** ******** (password was here)
on
1883
xblinds/all
xblinds-3160c0
512
Connecting to cecg_private
..................................
_________________________________________
**it never connects however  the ones that work look like this** 

----
eading stored values from EEPROM...
xBlinds
cecg_private
Xxxxxxx Xxxxxxx (password was here)
off    **_(note this whatever it is is off and the ones that don't work it is on)_**
192.168.16.68
1883

xblinds/3
xblinds-70d7e2

0
0
0
0
0
0

512

Connecting to cecg_private
___________________________________________________
**_Any info how to solve this would be appreciated_**

kp-bit, 2021-09-11:
Hi @mitchcecg, I would expect to see some default values in your output, which version of the binary are you using? Make sure you flash the latest one: https://github.com/kp-bit/xblinds/releases/tag/v0.8

mitchcecg, 2021-09-11:
Yes i am using that version

THANX
MITCH

kp-bit, 2021-09-12:
Well if some of them connect and others don't, it's probably not a code issue. More likely wifi antenna and/or distance. What brand of D1's are you using? Do you have a link for the exact model?

mitchcecg, 2021-09-12:
This is the one i got

ACEIRMC ESP8266 ESP-12 ESP-12F NodeMcu Mini D1 Module WeMos Lua 4M Bytes WLAN WiFi In... https://www.amazon.com/dp/B081PX9YFV/ref=cm_sw_r_sms_apip_ZEy2ccg6VDxgv

As i stated the ones that work have off it the screen shots i sent you and the ones that don’t have on in the screen shots but I have no idea what that means without the source code

THANX
MITCH

kp-bit, 2021-09-14:
Hi @mitchcecg, the on/off is whether the local AP is active (will give those texts a label in next version). I think your problem could be that you have saved the MQTT config without specifying a server, the "xblinds/3" indicates that you have changed the topic and saved, this is known to screw things up. Try resetting, just configure wifi and see if you get a connection, then add MQTT (if needed) with a valid IP address of a server.

closed 2021-09-14 by kp-bit

kp-bit, 2021-09-15:
Published v0.9 today, try and see if that helps you...

mitchcecg, 2021-09-17:
Still no Joy

***@***.***

[CECG_trans_Logo]THANX
Mitch Geier
President
Computer Emergency & Consulting Group, LLC.
***@***.******@***.***>
609-758-1111 Ext 1

Privileged-Confidential Information may be contained in this message. If you are not the addressee indicated in this message (or responsible for delivery of the message to such person), you may not copy or deliver this message to anyone. In such case, you should destroy this message and kindly notify the sender by reply email. Please advise immediately if you or your employer do not consent to Internet email for messages of this kind.

The sender believes that this E-mail and any attachments were free of any virus, worm, Trojan horse, and/or malicious code when sent. This message and its attachments could have been infected during  transmission. By reading the message and opening any attachments, the recipient accepts full responsibility for taking protective and remedial action about viruses and other defects. CECG LLC. and Mitch Geier are not liable for any loss or damage arising in any way from this message or its attachments.
P Please consider the environment before printing this email.

mitchcecg, 2021-09-17:
BTW this is the one that works

AP is off in the working ones

***@***.***
[CECG_trans_Logo]THANX
Mitch Geier
President
Computer Emergency & Consulting Group, LLC.
***@***.******@***.***>
609-758-1111 Ext 1

Privileged-Confidential Information may be contained in this message. If you are not the addressee indicated in this message (or responsible for delivery of the message to such person), you may not copy or deliver this message to anyone. In such case, you should destroy this message and kindly notify the sender by reply email. Please advise immediately if you or your employer do not consent to Internet email for messages of this kind.

The sender believes that this E-mail and any attachments were free of any virus, worm, Trojan horse, and/or malicious code when sent. This message and its attachments could have been infected during  transmission. By reading the message and opening any attachments, the recipient accepts full responsibility for taking protective and remedial action about viruses and other defects. CECG LLC. and Mitch Geier are not liable for any loss or damage arising in any way from this message or its attachments.
P Please consider the environment before printing this email.

kp-bit, 2021-09-18:
If AP is on at boot time, wifi isn't configured and saved properly. Are you testing all of them from the same location? Are you able to save the config if connecting to the AP and 4.3.2.1. Have you tried making a new (2.4 GHz only) SSID with a different password to see if that changes anything?

mitchcecg, 2021-09-18:
I created a new network with 2.4 GHz only still having same issue  it  is just about 4 feet from the wifi access point.

***@***.***

[CECG_trans_Logo]THANX
Mitch Geier
President
Computer Emergency & Consulting Group, LLC.
***@***.******@***.***>
609-758-1111 Ext 1

Privileged-Confidential Information may be contained in this message. If you are not the addressee indicated in this message (or responsible for delivery of the message to such person), you may not copy or deliver this message to anyone. In such case, you should destroy this message and kindly notify the sender by reply email. Please advise immediately if you or your employer do not consent to Internet email for messages of this kind.

The sender believes that this E-mail and any attachments were free of any virus, worm, Trojan horse, and/or malicious code when sent. This message and its attachments could have been infected during  transmission. By reading the message and opening any attachments, the recipient accepts full responsibility for taking protective and remedial action about viruses and other defects. CECG LLC. and Mitch Geier are not liable for any loss or damage arising in any way from this message or its attachments.
P Please consider the environment before printing this email.

mitchcecg, 2021-09-18:
Thanks for all the help. Finally found the issue it seems to be a windows dhcp issue I am working on a solution.

[CECG_trans_Logo]THANX
Mitch Geier
President
Computer Emergency & Consulting Group, LLC.
***@***.******@***.***>
609-758-1111 Ext 1

Privileged-Confidential Information may be contained in this message. If you are not the addressee indicated in this message (or responsible for delivery of the message to such person), you may not copy or deliver this message to anyone. In such case, you should destroy this message and kindly notify the sender by reply email. Please advise immediately if you or your employer do not consent to Internet email for messages of this kind.

The sender believes that this E-mail and any attachments were free of any virus, worm, Trojan horse, and/or malicious code when sent. This message and its attachments could have been infected during  transmission. By reading the message and opening any attachments, the recipient accepts full responsibility for taking protective and remedial action about viruses and other defects. CECG LLC. and Mitch Geier are not liable for any loss or damage arising in any way from this message or its attachments.
P Please consider the environment before printing this email.