kp-bit/xblinds #28: Xblinds gets disconnected and does not update MQTT status

denniskoenders, 2021-11-20:
Hi,
First of all, thank you for you project, I am loving it!.
But it seems that when the blind gets operated, the ESP gets disconnected after a short while with the MQTT message "xblinds-xxxxxx not connected".
The complete blind open or close procedure finishes correctly, but at the end I get the MQTT messagage "rebooted".
So the open or close status does not get updated in Home Assistant. Untill a short time ago, not a big problem, but now after HA was updated, the blind keeps on openening and closing.
I have tried several D1 mini's, all give the same result.
Is there something you can do in the Firmware to solve this issue?

Thanks very much.

With kind regards, Dennis

[image]

kp-bit, 2021-11-20:
I had the exact same problem after the latest HA update. You can use MQTT Explorer to clear out lingering commands in the MQTT queue and update to this v1.0beta .bin, that has fixed it for me. Expecting to release it in a couple of weeks...
[xblinds-v1.0b.zip](https://github.com/kp-bit/xblinds/files/7574601/xblinds-v1.0b.zip)

closed 2021-11-20 by kp-bit

denniskoenders, 2021-11-23:
Thanks very much.The screen works again, but still no correct status in Home Assistant.Will this also be fixed in a future Software update? Thanks again. Dennis  Verzonden vanuit Mail voor Windows Van: kp-bitVerzonden: zaterdag 20 november 2021 14:28Aan: kp-bit/xblindsCC: denniskoenders; AuthorOnderwerp: Re: [kp-bit/xblinds] Xblinds gets disconnected and does not update MQTT status (Issue #28) I had the exact same problem after the latest HA update. You can use MQTT Explorer to clear out lingering commands in the MQTT queue and update to this v1.0beta .bin, that has fixed it for me. Expecting to release it in a couple of weeks...xblinds-v1.0b.zip