kp-bit/xblinds #30: Suggestion: change the behavior of adjustment

Electr1fying, 2022-04-10:
Hi kp-bit!

Thanks for your software. I have a few suggestions which i'm giving their own topic so there won't be any garbage.

Is it possible to adjust the + and - behavior when setting up the blinds, or perhaps add another + and - for the bigger adjustment increments?

It is a pain to roll out a 160cm blind for setting its position with such small increments. I think it would be better to add (or adjust it to) a +, - and STOP.

Thanks in advance!

kp-bit, 2022-04-13:
Hi @Electr1fying, I believe this is already covered by the "change step size" under Experimental in Setup. Use the feature carefully!

closed 2022-04-13 by kp-bit

Electr1fying, 2022-04-13:
Hi kp-bit,

Check thanks. I didn't know this feature was meant for the adjustment and to change the +/- roller adjustment behaviour!