kp-bit/xblinds #31: Suggestion: MQTT in JSON and/or implement Home Assistant Auto-Discovery

Electr1fying, 2022-04-10:
Hi!

Might it be possible to format xblinds MQTT Messages into JSON, and/or implement Home Assistant Auto-Discovery?

I'm using Domoticz myself, but a recent Domoticz release has support For HASS Auto-Discovery, which should help getting it into Domoticz as well i reckon.

I think part of the problem is; i've sniffed xblinds's MQTT traffic and it does not seem to be JSON formatted. This is the MQTT formatting Domoticz supports and expects, but i reckon HASS Auto-Discovery should help as well with that.

Further i read that it might be possible to "convert" xblind's MQTT messages to a JSON format with the help of Node-Red, but i'm not to handy with it and i rather don't want to use tricks into getting it to work via some sort of translation layer.

Thanks in advance!

kp-bit, 2022-04-13:
Hi @Electr1fying, I'll consider adding autodiscover in a future version. v1.0 will be released without it, as I don't have time for a testing cycle right now. As for JSON formatting, I really have no way of testing it, as I use HA myself (and focus on supporting HA), so you will have to mess with that yourself when source is released.

Electr1fying, 2022-04-13:
Hi kp-bit!

Thanks. Will take a look at the source once released (hopefully soon).

In which format will you release the source? .ino for Arduino IDE?

kp-bit, 2022-04-13:
Yes, for the time being it will be .ino + libaries...

Electr1fying, 2022-04-13:
Super. Thanks alot for your work and effort!

kp-bit, 2022-04-15:
Source code has now been published, I've added autodiscover to the planned work for next version