kp-bit/xblinds #37: MQTT HA Problem

raphhorn, 2023-10-15:
everytime Home assitant reboots i have to type in the Mqtt paswwort on the Web Interface of each blind then it works but its everytime

kp-bit, 2023-10-16:
I need a little more information... Are you using the exact D1 mini listed in the BOM? Which MQTT broker are you using and is it updated? What version of HA? Have you tried reflashing the D1 with latest build?

kp-bit, 2023-10-19:
Sounds like there's no "EEPROM" on your modules - do you have a normal ESP8266 to test with? I've had 4 D1's running in production for more than 2 years and never seen this issue. And if it was a general issue, I'd have heard about it before...

raphhorn, 2023-10-19:
You think that the D1 Mini forgot the passwort but the problems appears when i start HA new so the D1 Mini has constant power and it works until i restart HA

kp-bit, 2023-10-19:
What if you just restart the D1, after HA has restarted? Does it reconenct then?

raphhorn, 2023-10-19:
> What if you just restart the D1, after HA has restarted? Does it reconenct then?

No it works only if i type in the mqtt passwort everything else is remebered

kp-bit, 2023-10-19:
So the password is actually remembered, it's just not exposed in the textbox on the webpage (same on the wifi page). Could you try presssing "Reboot ESP" on the "Setup" page, instead of typing the password and see if it reconnects?

raphhorn, 2023-10-19:
> So the password is actually remembered, it's just not exposed in the textbox on the webpage (same on the wifi page). Could you try presssing "Reboot ESP" on the "Setup" page, instead of typing the password and see if it reconnects?

Yeah i know that it is not exposed 
I tried alredy reboot the esp or unplug it for a couple seconds

raphhorn, 2023-10-19:
I check all my setting and your Troubleshooting tipps and my mosquitto settings and let you know if it still there

raphhorn, 2023-10-19:
everytime the mosquitto broker restarts the problem appears

kp-bit, 2023-10-20:
It should restart eventually on lost communication with the broker (if detected properly). I saw something similar in HA 2021.11.x where I had to toggle power to the blinds, so you might want to make sure you're on the latest HA. Alternatively, you can play around with MQTT KeepAlive (line 470 in the code) if you're up for recompiling it. Don't think there's much I can do here, so will close the issue.

closed 2023-10-20 by kp-bit