kp-bit/xblinds #38: not an Issue but an Idea

raphhorn, 2023-12-13:
I'm sorry I don't know how to direct message you, but I have an Idea.
I use your firmware for my own blinds configuration and there are the complete electronics are on the window and I thought it would be nice to attach a switch or reed contact to the window connected with the d1 mini which is sent via Mqtt to home assistant to display the status of the window if its open or closed. Unfortunately, I didn't know anything about coding, so I thought you maybe can add this feature. 

I would also pay for a coffee :)

My Automated Blinds Version with your Firmware: https://www.printables.com/de/model/383885-automated-blinds

kp-bit, 2023-12-14:
Hi @raphhorn, thanks for the suggestion. While it should be fairly easy to implement sensors in the code and send status via MQTT, it's pretty far from the core functionality of xblinds (so is your creative way of using it, but good job!) - so I'm gonna have to decline implementing other functions. I can recommend Aqara magnetic door/window sensors.

closed 2023-12-14 by kp-bit

raphhorn, 2023-12-14:
Hi,
                Thanks for the quick answer I have already monitored my windows with door sensors but I thought it would be a good solution to avoid the battery problem because the d1 mini is constantly supplied and the hardware would be available.And on top of that, you could use such a contact or executed as a button also as a button in home assistant to open all blinds, for exampleBut I can survive if they don't want to implement this for the reasons mentioned.ThanksGesendet mit der mobilen Mail AppAm 14.12.23 um 07:57 schrieb kp-bit
                
            
            
                
                    Von: "kp-bit" ***@***.***>Datum: 14. Dezember 2023An: "kp-bit/xblinds" ***@***.***>Cc: "raphhorn" ***@***.***>,"Mention" ***@***.***>Betreff: Re: [kp-bit/xblinds] not an Issue but an Idea (Issue #38)
                
                
Hi @raphhorn, thanks for the suggestion. While it should be fairly easy to implement sensors in the code and send status via MQTT, it's pretty far from the core functionality of xblinds (so is your creative way of using it, but good job!) - so I'm gonna have to decline implementing other functions. I can recommend Aqara magnetic door/window sensors.